nutdotnet/WinNUT-Client #145: WinNUT incorrectly displays a "Not Connected" notification on Resume

yoyoma2, 2024-03-31:
- [x] Searched to see if this issue exists
- [x] Completed below form
- [x] Attached debug log file
- [x] Copy output in File -> UPS Variable

**WinNUT Version: ** v2.2.8719.24624
**Windows OS Version: ** Windows 10 Pro 22H2

**Describe the bug**
When Windows 10 resumes from hibernation, WinNUT notifications falsely show that it connected then disconnected. It appears that this is a false alarm as WinNUT seems to be connected. The logs below show that the disconnection happens first and the connection happens later albeit the timestamp is identical which may explain the misordered notifications.

**To Reproduce**
Steps to reproduce the behavior:
1. Configure Win10 to hibernate and wait for it to happen
2. Come back hours later and wake up Windows
3. Look at Windows notifications in the image (most recent notification is at the top):

**Expected behavior**
The "Connected" notification should appear last since it happened last. Ideally the "Not Connected" notification should appear much earlier when the sleep state is initiated. If I manually shutdown Windows, the "Not Connected" notification is tagged with the time the shutdown was initiated.

**Screenshots**
[image]

**Additional context**
I tried to hibernate manually and immediately resume and that didn't reproduce the problem. I guess the long hibernating time leaves the sockets in a different state than the manual immediate hibernate/resume.

[WinNUT-Client-2024-03-28.log](https://github.com/nutdotnet/WinNUT-Client/files/14816159/WinNUT-Client-2024-03-28.log)
[upsvars.txt](https://github.com/nutdotnet/WinNUT-Client/files/14816172/upsvars.txt)

gbakeman, 2024-04-10:
I definitely noticed some spammy notifications too, usually when resuming a session. I'll look into this.

gbakeman, 2024-07-18:
@yoyoma2 I'm about to put up another pre-release with some connection and operational fixes included. I'd like to hear from you if you think I've improved this problem or not. Thanks!

yoyoma2, 2024-07-19:
> @yoyoma2 I'm about to put up another pre-release with some connection and operational fixes included. I'd like to hear from you if you think I've improved this problem or not. Thanks!

Sure I'll try a July Pre-Release when I see it posted.

gbakeman, 2024-07-24:
The new pre-release is up - please let me know what you think!

yoyoma2, 2024-07-24:
Installed pre-release v2.3.8970 (when running v2.3.8971 appears in the about box however).

The old settings were imported (nice!) but I had to change one calibration setting by 1V then change it back or some of the dials were showing default ranges instead of my custom settings.

As for better suspend/resume support I'll need to test that a little later when the system isn't in use.

gbakeman, 2024-07-24:
Good catch on gauge calibrations, I noticed that too at some point or another but forgot to open an issue about it. Feel free to subscribe to #170 if that's something you're interested in seeing fixed.

Thanks for checking in and in advance for testing this further!

yoyoma2, 2024-07-26:
After coming out of hibernation there were no notifications and the icon in the taskbar had some red on it. Going to Connection->Reconnect fixed everything.
I'll try to reproduce with some logs next time.

gbakeman, 2024-07-26:
I'm actually curious if you can test with the build [here](https://github.com/nutdotnet/WinNUT-Client/actions/runs/10100965530) (once you get the chance) since I've addressed more notification and connection issues since the pre-release. Thanks!