nutdotnet/WinNUT-Client #150: Huawei UPS missing load display

gbakeman, 2024-04-28:
@MaxMadcc 

<details>

<summary>UPS variables</summary>

```HUAWEI01 (Huawei/UPS2000A/V2R1C1SPC40)
battery.capacity (Battery capacity (Ah)) : 9
battery.charge (Battery charge (percent of full)) : 100.0
battery.charger.status (Battery charger status) : charging
battery.packs (Number of battery packs) : 4
battery.runtime (Battery runtime (seconds)) : 2418
battery.voltage (Battery voltage (V)) : 54.4
device.mfr (Description unavailable) : Huawei
device.model (Description unavailable) : UPS2000A
device.serial (Description unavailable) : 2102290503HGJ7000056
device.type (Description unavailable) : ups
driver.debug (Current debug verbosity level of the driver program) : 1
driver.flag.allow_killpower (Safety flip-switch to allow the driver daemon to send UPS shutdown command (accessible via driver.killpower)) : 0
driver.name (Driver name) : huawei-ups2000
driver.parameter.pollinterval (Description unavailable) : 3
driver.parameter.port (Description unavailable) : /dev/ttySVA100
driver.parameter.synchronous (Description unavailable) : no
driver.state (Description unavailable) : updateinfo
driver.version (Driver version - NUT release) : 2.8.1
driver.version.internal (Internal driver version) : 0.05
input.bypass.frequency (Description unavailable) : 50.0
input.bypass.voltage (Description unavailable) : 232.6
input.frequency (Input line frequency (Hz)) : 50.0
input.voltage (Input voltage (V)) : 232.6
output.current (Output current (A)) : 1.7
output.frequency (Output frequency (Hz)) : 50.0
output.power (Description unavailable) : 300.0
output.realpower (Description unavailable) : 200.0
output.voltage (Output voltage (V)) : 219.5
ups.beeper.status (UPS beeper status) : enabled
ups.delay.reboot (Interval to wait before rebooting the UPS (seconds)) : 60
ups.delay.shutdown (Interval to wait after shutdown with delay command (seconds)) : 60
ups.delay.start (Interval to wait before (re)starting the load (seconds)) : 60
ups.firmware (UPS firmware) : V2R1C1SPC40
ups.firmware.aux (Auxiliary device firmware) : P1.0-D1.0
ups.load (Load on UPS (percent of full)) : 18.0
ups.mfr (UPS manufacturer) : Huawei
ups.model (UPS model) : UPS2000A
ups.power.nominal (UPS power rating (VA)) : 2000
ups.serial (UPS serial number) : 21
ups.status (UPS status) : OL CHRG
ups.temperature (UPS temperature (degrees C)) : 28.4
ups.timer.reboot (Time before the load will be rebooted (seconds)) : -1
ups.timer.shutdown (Time before the load will be shutdown (seconds)) : -1
ups.timer.start (Time before the load will be started (seconds)) : -1
ups.type (UPS type) : online
```

</details>

> The output power should be calculated according to the following.
P = I*V 
It appears that my UPS variables include **output.realpower** and **output.power**, but the  the load wattage graph show 0W and percentage of real power.

<details>

<summary>Log (part)</summary>

```2024/4/28 23:06:08 [30336, Nut_Socket]: Attempting authentication...
2024/4/28 23:06:08 [30336, Nut_Socket]: NUT server reports VER: 2.8.1 NETVER: 
2024/4/28 23:06:08 [30336, UPS_Device]: Retrieving basic UPS product information...
2024/4/28 23:06:08 [30336, UPS_Device]: Determining best method to calculate power usage...
2024/4/28 23:06:08 [30336, UPS_Device]: ups.realpower is not supported by server.
2024/4/28 23:06:08 [30336, UPS_Device]: ups.realpower.nominal is not supported by server.
2024/4/28 23:06:08 [30336, UPS_Device]: input.current.nominal is not supported by server.
2024/4/28 23:06:08 [30336, UPS_Device]: Unable to find a suitable method to calculate power usage.
2024/4/28 23:06:08 [30336, UPS_Device]: output.frequency.nominal is not supported by server.
2024/4/28 23:06:08 [30336, UPS_Device]: Apply Fallback Value when retrieving output.frequency.nominal
2024/4/28 23:06:08 [30336, UPS_Device]: Completed retrieval of basic UPS product information.
2024/4/28 23:06:08 [30336, WinNUT]: Connection to Nut Host Established
2024/4/28 23:06:09 [30336, UPS_Device]: UPS statuses have CHANGED...
2024/4/28 23:06:09 [30336, UPS_Device]: Current statuses: OL,CHRG
2024/4/28 23:06:59 [30336, UPS_Device]: UPS device updating status is now [False]
2024/4/28 23:06:59 [30336, UPS_Device]: UPS device updating status is now [True]
2024/4/28 23:28:29 [30336, UPS_Device]: UPS device updating status is now [False]
2024/4/28 23:28:29 [30336, UPS_Device]: UPS device updating status is now [True]
2024/4/28 23:37:27 [30336, Pref_Gui]: Opened UI window to log location.
2024/4/28 23:37:39 [30336, WinNUT]: Disconnected from Nut Host
2024/4/28 23:37:39 [30336, WinNUT]: Client UPS_Connect subroutine beginning.
```

</details>

##Notes
We seem to be missing `output.current`, `output.voltage`, and `output.realpower` for load calculation.

gbakeman, 2024-04-28:
@MaxMadcc I'm noticing that the calculation for power based on current and voltage (219.5*1.7=373.15) is quite a lot different than your output.realpower reading. Can you confirm which if any of the variables/calculations represents the true power draw of your UPS?

MaxMadcc, 2024-05-04:
According to the [official manual](https://support.huawei.com/enterprise/zh/doc/DOC1000110695) and [nut huawei-ups2000 driver](https://github.com/networkupstools/nut/blob/ff045c089b362085ec0abd651da5e9216c49e281/drivers/huawei-ups2000.c#L552) I think "**`output.realpower`**" is appropriate for reading values.
I think that in certain UPS systems, "output.realpower" may not be readable, so use P=I * V * 0.8 (note: PF).

gbakeman, 2024-05-04:
Thank you for finding that information! As I'm understanding it for now, it seems like `output.power` doesn't really have any use to us so I think I'll ignore that unless you have other ideas.

I'd like to propose a few tests if you wouldn't mind running them for me. I'm going to work on the power calculation method first so it supports your UPS, then I'd like you to give that build a try for me and report how it works. After that, I'll implement the direct reading from `output.realpower` as that seems to be the most convenient and accurate measurement available to us. Then I'd like you to test that and let me know how it works, then I'll have a pre-release up.

Let me know what you think about this.

gbakeman, 2024-05-04:
Just put up a new debug build [here](https://github.com/nutdotnet/WinNUT-Client/actions/runs/8953178531?pr=151), please download and test when convenient. This should cause WinNUT to calculate power based on your UPS's output voltage and current only. More direct variable reading will come in the future.

MaxMadcc, 2024-05-05:
[image]
It seems to need round 1 or 2 number to the decimal places.

gbakeman, 2024-05-05:
🤦

I'll try using NUT/upsd to put up a virtual UPS with your variables next time so I catch simple mistakes like this! I think the extra decimal places fix may be best applied at the gauge its self. I'm tempted to wrap that work into #136, especially since they too are missing the battery.voltage reading.

But in the meantime, I'll work on applying the next sub-feature here which is reading directly from `output.realpower`. Do you feel the current reading is fairly accurate then? Your `realpower` last time was ~200W and the `power` variable was around ~300W, so I wonder if you may have just been using more power when taking this picture.

gbakeman, 2024-05-06:
I was able to emulate your UPS based on the variables you gave me earlier, so hopefully I catch issues in the future!

[image]

I'll look into rounding that Load calculation and implement the next step.

gbakeman, 2024-05-06:
[Another build is up now](https://github.com/nutdotnet/WinNUT-Client/actions/runs/8971300017?pr=151), this time with direct reading of `output.realpower` as well as correcting the decimal places. Please give it a try and let me know how it goes. I'll take a look at your translation PR then have a pre-release up soon.

PR #151 (2024-05-06): Power value calculation updates (specifically for Huawei UPS)
- Added output current as new `UPS_Value`
- Renamed a few `PowerMethod` enums to be more descriptive of what they do
- Trying a new well-known variable initialization method that utilizes nullable values. Variable is left null if there's a problem retrieving it from the NUT server, then it will not be updated in the future. Only applying this to `output.current` since its use is limited to power calculation only. Applying this to other variables will likely require greater structural changes throughout WinNUT.
- Added power calculat [... cut]
closed 2024-05-06 by gbakeman