nutdotnet/WinNUT-Client #157: Remaing Time Unavalible Pc Shuts down imediatly

FigJam23, 2024-05-26:
**Please make sure to fill out this form completely, and attach your program logs (set to Debug setting). You can delete this line.**

- [ ] Searched to see if this issue exists
- [ ] Completed below form
- [ ] Attached debug log file
- [ ] Copy output in File -> UPS Variable

**WinNUT Version: 2.3.8892.2** 
**Windows OS Version: w10 ** 

**Describe the bug**
Remaining Time Unknowen on my Powersheild rt2000 and when power disconnects from ups winnut try's shutting down immediately 

running on RPI using upsc powershield@192.168.1.200
```
######### Power Sheild UPS ###################
battery.charge: 100
battery.voltage: 54.70
battery.voltage.high: 52.00
battery.voltage.low: 41.60
battery.voltage.nominal: 48.0
device.mfr:
device.model: LIHVX2K0
device.type: ups
driver.name: blazer_usb
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.productid: 5161
driver.parameter.synchronous: auto
driver.parameter.vendorid: 0665
driver.version: 2.8.0
driver.version.internal: 0.14
driver.version.usb: libusb-1.0.26 (API: 0x1000109)
input.current.nominal: 8.0
input.frequency: 50.0
input.frequency.nominal: 50
input.voltage: 231.3
input.voltage.fault: 231.2
input.voltage.nominal: 240
output.voltage: 231.2
ups.beeper.status: enabled
ups.delay.shutdown: 30
ups.delay.start: 180
ups.firmware: 00303.06
ups.load: 2
ups.mfr:
ups.model: LIHVX2K0
ups.productid: 5161
ups.status: OL
ups.temperature: 16.5
ups.type: offline / line interactive
ups.vendorid: 0665
admin@MikesNetMon:~ $
```
Using same cmd on my EATON  ups seems to work

```
################ Eaton UPS ###################
admin@MikesNetMon:~ $ upsc eaton@192.168.1.200
Init SSL without certificate database
battery.charge: 100
battery.runtime: 2637
battery.type: PbAc
device.mfr: EATON
device.model: 5E 850i
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.synchronous: auto
driver.version: 2.8.0
driver.version.data: MGE HID 1.46
driver.version.internal: 0.47
driver.version.usb: libusb-1.0.26 (API: 0x1000109)
input.voltage: 234.0
outlet.1.status: on
outlet.desc: Main Outlet
outlet.id: 1
outlet.switchable: no
output.frequency: 49.9
output.frequency.nominal: 50
output.voltage: 232.0
output.voltage.nominal: 230
ups.beeper.status: enabled
ups.delay.shutdown: 20
ups.firmware: 03.08.0018
ups.load: 9
ups.mfr: EATON
ups.model: 5E 850i
ups.power.nominal: 850
ups.productid: ffff
ups.realpower: 61
ups.start.battery: yes
ups.status: OL
ups.timer.shutdown: -1
ups.vendorid: 0463
```

so my Eaton ups works but the powersheild ups does not seem to its likely due to missing this battery.runtime: value. and the issue caused on Winnut shutting down my pc almost straight away is a problem how can I get this to work or get the battery.runtime:  status from my powersheild ups. would anyone have any idea on  how to resolve this or stop Winnut shutting my pc down cant see a option to disable the shutdown either

FigJam23, 2024-05-26:
[image]

FigJam23, 2024-05-26:
ok i fixed my issue changed my usb driver from blazer_usb to nutdrv_qx and now have ``` admin@MikesNetMon:~ $ upsc powershield@192.168.1.200
Init SSL without certificate database
battery.charge: 94
battery.energysave: no
battery.packs: 1
battery.protection: yes
battery.runtime: 21180
battery.voltage: 54.70
battery.voltage.nominal: 48.0
device.model: LIHVX2K0
device.serial: 81201806100137
device.type: ups
driver.name: nutdrv_qx
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.productid: 5161
driver.parameter.synchronous: auto
driver.parameter.vendorid: 0665
driver.version: 2.8.0
driver.version.data: Voltronic 0.06
driver.version.internal: 0.32
driver.version.usb: libusb-1.0.26 (API: 0x1000109)
input.current.nominal: 8.0
input.frequency: 50.1
input.frequency.nominal: 50.0
input.phases: 1
input.voltage: 231.4
input.voltage.nominal: 240.0
outlet.0.switchable: no
output.current: 0.2
output.current.nominal: 8
output.frequency: 50.1
output.frequency.nominal: 50.0
output.phases: 1
output.power.maximum.percent: 2
output.power.minimum.percent: 2
output.powerfactor: 0.8
output.voltage: 231.5
output.voltage.nominal: 240.0
ups.beeper.status: enabled
ups.delay.shutdown: 30
ups.delay.start: 180
ups.firmware: 00303.06
ups.firmware.aux: P00
ups.load: 2
ups.power.nominal: 2000
ups.productid: 5161
ups.start.auto: yes
ups.start.battery: yes
ups.status: OL
ups.temperature: 23.3
ups.type: line-interactive
ups.vendorid: 0665
admin@MikesNetMon:~ $
```

gbakeman, 2024-05-26:
Thank you for posting back!

closed 2024-05-26 by gbakeman