nutdotnet/WinNUT-Client #162: WinNUT cannot reconnect automatically after Windows wakes up from sleep.

timmyhbk, 2024-06-27:
- [ V ] Searched to see if this issue exists
- [ V ] Completed below form
- [ V ] Attached debug log file
- [ V ] Copy output in File -> UPS Variable

**WinNUT Version:**  
2.3.8892.21649
**Windows OS Version:** 
Windows 11 23H2 22631.3737

**Describe the bug**
When Windows resumes from sleep, WinNUT cannot connect.
Log stay on Try Reconnect 1 / 30. ()
Need to manually reconnect to resume the connection.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect WinNUT to the NUT server.
2. Put Windows to sleep.
3. Wake up Windows.
4. See error.

**Expected behavior**
WinNUT can automatically reconnect when Windows wakes up from sleep.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
1. If it's a reboot. WinNUT can reconnect normally.

**Log** (I don't know how to get it to generate an English Log, so I've put an English translation below the text.)
```
2024/6/27 Thursday 下午 01:59:59 [2820, WinNUT]: PowerModeChangedEvent: Suspend
2024/6/27 Thursday 下午 02:00:00 [2820, UPS_Device]: Enter Retrieve_UPS_Datas
2024/6/27 Thursday 下午 02:06:38 [2820, UPS_Device]: Something went wrong in Retrieve_UPS_Datas:
2024/6/27 Thursday 下午 02:06:38 [2820, UPS_Device]: System.IO.IOException thrown in System
Message: 無法從傳輸連接讀取資料: 連線已被您主機上的軟體中止。。
(editor's notes: Unable to read data from the transport connection: The connection has been aborted by the software on your host.)
   At System.Net.Sockets.NetworkStream.Read(Byte[] buffer, Int32 offset, Int32 size)
   At System.IO.StreamReader.ReadBuffer()
   At System.IO.StreamReader.ReadLine()
   At WinNUT_Client_Common.Nut_Socket.Query_Data(String Query_Msg) At C:\Users\Glen\Repositories\nutdotnet\WinNUT-Client\WinNUT_V2\WinNUT-Client_Common\Nut_Socket.vb: Line 237
   At WinNUT_Client_Common.UPS_Device.GetUPSVar(String varName, Object Fallback_value, Boolean recursing) At C:\Users\Glen\Repositories\nutdotnet\WinNUT-Client\WinNUT_V2\WinNUT-Client_Common\UPS_Device.vb: Line 442
   At WinNUT_Client_Common.UPS_Device.Retrieve_UPS_Datas(Object sender, EventArgs e) At C:\Users\Glen\Repositories\nutdotnet\WinNUT-Client\WinNUT_V2\WinNUT-Client_Common\UPS_Device.vb: Line 325

2024/6/27 Thursday 下午 02:06:38 [2820, UPS_Device]: Inner exception present:
2024/6/27 Thursday 下午 02:06:38 [2820, IOException]: System.Net.Sockets.SocketException thrown in System
Message: 連線已被您主機上的軟體中止。
(editor's notes: The connection has been aborted by the software on your host.)
   At System.Net.Sockets.Socket.Receive(Byte[] buffer, Int32 offset, Int32 size, SocketFlags socketFlags)
   At System.Net.Sockets.NetworkStream.Read(Byte[] buffer, Int32 offset, Int32 size)

2024/6/27 Thursday 下午 02:06:38 [2820, Logger]: Exception report complete.
2024/6/27 Thursday 下午 02:06:38 [2820, Logger]: Exception report complete.
2024/6/27 Thursday 下午 02:06:38 [2820, UPS_Device]: Processing request to disconnect...
2024/6/27 Thursday 下午 02:06:38 [2820, WinNUT]: Initializing all display values and configurations.
2024/6/27 Thursday 下午 02:06:39 [2820, WinNUT]: Update Icon
2024/6/27 Thursday 下午 02:06:39 [2820, WinNUT]: Status Icon Changed
2024/6/27 Thursday 下午 02:06:39 [2820, WinNUT]: New Icon Value For Systray : 1216
2024/6/27 Thursday 下午 02:06:39 [2820, WinNUT]: New Icon Value For Gui : 1216
2024/6/27 Thursday 下午 02:06:39 [2820, WinNUT]: NotifyIcon Text => 
WinNUT Client - 2.3
Not Connected
2024/6/27 Thursday 下午 02:06:39 [2820, WinNUT]: Battery Status => Unknown
2024/6/27 Thursday 下午 02:06:39 [2820, WinNUT]: Disconnected from Nut Host
2024/6/27 Thursday 下午 02:06:39 [2820, String]: New Log to CB_Current Log : Disconnected from Nut Host
2024/6/27 Thursday 下午 02:06:39 [2820, WinNUT]: PowerModeChangedEvent: Resume
2024/6/27 Thursday 下午 02:06:39 [2820, WinNUT]: Restarting WinNUT after waking up from Windows
2024/6/27 Thursday 下午 02:06:39 [2820, String]: New Log to CB_Current Log : Restarting WinNUT after waking up from Windows.
[... 432 lines cut ...]
driver.parameter.synchronous (Description unavailable) : no
driver.version (Description unavailable) : DSM7-2-1-NewModel-repack-64570-230831
driver.version.data (Description unavailable) : CyberPower HID 0.4
driver.version.internal (Description unavailable) : 0.41
input.transfer.high (Description unavailable) : 135
input.transfer.low (Description unavailable) : 100
input.voltage (Description unavailable) : 110.0
input.voltage.nominal (Description unavailable) : 0
output.voltage (Description unavailable) : 110.0
ups.beeper.status (Description unavailable) : enabled
ups.delay.shutdown (Description unavailable) : 20
ups.delay.start (Description unavailable) : 30
ups.load (Description unavailable) : 21
ups.mfr (Description unavailable) : CPS
ups.model (Description unavailable) : CP650HGaTW
ups.productid (Description unavailable) : 0501
ups.realpower.nominal (Description unavailable) : 375
ups.status (Description unavailable) : OL
ups.test.result (Description unavailable) : Done and passed
ups.timer.shutdown (Description unavailable) : -60
ups.timer.start (Description unavailable) : -60
ups.vendorid (Description unavailable) : 0764
```

**Thanks for your help.**

gbakeman, 2024-06-28:
Hi @timmyhbk ,

I have to wonder if this is related to #145. WinNUT currently does not correctly handle the OS going from active -> suspend -> active states, so the next time system operation is resumed, it will continue to try exchanging data over a broken socket connection. This will almost immediately result in an error, which it reports, then tries reconnecting again. From what I can see in the log you posted, that seems to be the case here. You suspended your system at 01:59:59, then resumed around 02:06. WinNUT discovers the broken socket and goes through a reconnect procedure. After dealing with a strange response from the Synology DSM (fairly typical), it then seems to resume reading data from the NUT server normally.

It looks like WinNUT was actually reconnected successfully after resuming from sleep. Please let me know if you think issue #145 is more closer to what you're experiencing, otherwise we'll need to troubleshoot this a little more.

timmyhbk, 2024-06-29:
Hi @gbakeman ,

After testing for a while I did notice that occasionally the connection would be restored. But not always.

When it doesn't reconnect, the GUI doesn't change, even if the UPS is disconnected. Clicking on the UPS Variable brings up Error Encountered.

[image]

I tried logging different logs. hope this helps.

```
2024/6/29 星期六 下午 05:27:42 [19320, WinNUT]: Update Icon
2024/6/29 星期六 下午 05:33:04 [19320, WinNUT]: PowerModeChangedEvent: Suspend
2024/6/29 星期六 下午 05:33:04 [19320, WinNUT]: PowerModeChangedEvent: Resume
2024/6/29 星期六 下午 05:33:04 [19320, WinNUT]: Restarting WinNUT after waking up from Windows
2024/6/29 星期六 下午 05:33:04 [19320, String]: New Log to CB_Current Log : 從睡眠喚醒 Windows 後重新啟動 WinNUT。 (edit: Restarting WinNUT after waking Windows from sleep)
2024/6/29 星期六 下午 05:33:06 [19320, WinNUT]: Client UPS_Connect subroutine beginning.
2024/6/29 星期六 下午 05:33:06 [19320, UPS_Device]: Beginning connection: upsmon@192.168.1.101:3493, Name: ups [AutoReconnect]
2024/6/29 星期六 下午 05:33:06 [19320, Nut_Socket]: Attempting TCP socket connection to 192.168.1.101:3493...
2024/6/29 星期六 下午 05:33:06 [19320, Nut_Socket]: Connection established and streams ready for 192.168.1.101:3493
2024/6/29 星期六 下午 05:33:06 [19320, Nut_Socket]: Attempting authentication...
2024/6/29 星期六 下午 05:33:06 [19320, Nut_Socket]: Error while attempting to log in: ACCESSDENIED (ERR ACCESS-DENIED)
Query: USERNAME upsmon
2024/6/29 星期六 下午 05:33:06 [19320, UPS_Device]: NUT protocol error encoutnered:
WinNUT_Client_Common.NutException: ACCESSDENIED (ERR ACCESS-DENIED)
Query: VER
   At WinNUT_Client_Common.Nut_Socket.Query_Data(String Query_Msg)
   At WinNUT_Client_Common.Nut_Socket.Connect()
   At WinNUT_Client_Common.UPS_Device.Connect_UPS(Object retryOnConnFailure)
2024/6/29 星期六 下午 05:33:07 [19320, UPS_Device]: Enter Retrieve_UPS_Datas
2024/6/29 星期六 下午 05:33:07 [19320, UPS_Device]: Something went wrong in Retrieve_UPS_Datas: System.IO.IOException: 無法寫入資料至傳輸連接: 遠端主機已強制關閉一個現存的連線。。 ---> System.Net.Sockets.SocketException: 遠端主機已強制關閉一個現存的連線。
(Edit: Unable to write data to the transport connection: The remote host has forcibly closed an existing connection. // The remote host has forced an existing connection to close.)
   At System.Net.Sockets.Socket.Send(Byte[] buffer, Int32 offset, Int32 size, SocketFlags socketFlags)
   At System.Net.Sockets.NetworkStream.Write(Byte[] buffer, Int32 offset, Int32 size)
   --- 內部例外狀況堆疊追蹤的結尾 ---
   (Edit: End of Internal Exception Stack Tracking)
   At System.Net.Sockets.NetworkStream.Write(Byte[] buffer, Int32 offset, Int32 size)
   At System.IO.StreamWriter.Flush(Boolean flushStream, Boolean flushEncoder)
   At System.IO.StreamWriter.Flush()
   At WinNUT_Client_Common.Nut_Socket.Query_Data(String Query_Msg)
   At WinNUT_Client_Common.UPS_Device.GetUPSVar(String varName, Object Fallback_value, Boolean recursing)
   At WinNUT_Client_Common.UPS_Device.Retrieve_UPS_Datas()
2024/6/29 星期六 下午 05:33:07 [19320, UPS_Device]: Processing request to disconnect...
2024/6/29 星期六 下午 05:33:07 [19320, UPS_Device]: Socket has reported a Broken event.
2024/6/29 星期六 下午 05:33:07 [19320, UPS_Device]: Reconnection Process Started
2024/6/29 星期六 下午 05:33:12 [19320, UPS_Device]: Try Reconnect 1 / 30
2024/6/29 星期六 下午 05:33:12 [19320, String]: New Log to CB_Current Log : Try Reconnect 1 / 30
2024/6/29 星期六 下午 05:33:12 [19320, UPS_Device]: Beginning connection: upsmon@192.168.1.101:3493, Name: ups [AutoReconnect]
2024/6/29 星期六 下午 05:33:12 [19320, Nut_Socket]: Attempting TCP socket connection to 192.168.1.101:3493...
2024/6/29 星期六 下午 05:33:12 [19320, Nut_Socket]: Connection established and streams ready for 192.168.1.101:3493
2024/6/29 星期六 下午 05:33:12 [19320, Nut_Socket]: Attempting authentication...
2024/6/29 星期六 下午 05:33:12 [19320, Nut_Socket]: Error while attempting to log in: INVALIDARGUMENT (ERR INVALID-ARGUMENT)
Query: LOGIN
2024/6/29 星期六 下午 05:33:12 [19320, Nut_Socket]: NUT server reports VER: DSM7-2-1-NewModel-repack-64570-230831 NETVER: 1.2
2024/6/29 星期六 下午 05:33:12 [19320, UPS_Device]: ups.serial is not supported by server.
2024/6/29 星期六 下午 05:33:12 [19320, UPS_Device]: Apply Fallback Value when retrieving ups.serial
2024/6/29 星期六 下午 05:33:12 [19320, UPS_Device]: ups.firmware is not supported by server.
2024/6/29 星期六 下午 05:33:12 [19320, UPS_Device]: Apply Fallback Value when retrieving ups.firmware
2024/6/29 星期六 下午 05:33:12 [19320, UPS_Device]: ups.realpower is not supported by server.
2024/6/29 星期六 下午 05:33:12 [19320, UPS_Device]: Unhandled error while getting ups.realpower
2024/6/29 星期六 下午 05:33:12 [19320, UPS_Device]: Using NominalPowerCalc method to calculate power usage.
2024/6/29 星期六 下午 05:33:12 [19320, UPS_Device]: battery.capacity is not supported by server.
2024/6/29 星期六 下午 05:33:12 [19320, UPS_Device]: Apply Fallback Value when retrieving battery.capacity
2024/6/29 星期六 下午 05:33:12 [19320, UPS_Device]: output.frequency.nominal is not supported by server.
2024/6/29 星期六 下午 05:33:12 [19320, UPS_Device]: Apply Fallback Value when retrieving output.frequency.nominal
2024/6/29 星期六 下午 05:33:12 [19320, UPS_Device]: Nut Host Reconnected
2024/6/29 星期六 下午 05:33:15 [19320, WinNUT]: Restore Main Gui On Mouse Click Notify Icon
2024/6/29 星期六 下午 05:33:15 [19320, WinNUT]: Main Gui Has Focus
2024/6/29 星期六 下午 05:33:15 [19320, WinNUT]: Update Icon
2024/6/29 星期六 下午 05:33:15 [19320, WinNUT]: Main Gui Lose Focus
[... 39 lines cut ...]
[... cut ...]
2024/6/29 星期六 下午 05:33:47 [19320, UPS_Device]: Apply Fallback Value when retrieving input.frequency
2024/6/29 星期六 下午 05:33:47 [19320, UPS_Device]: UPS statuses have not changed since last update, skipping.
2024/6/29 星期六 下午 05:33:48 [19320, WinNUT]: Main Gui Has Focus
2024/6/29 星期六 下午 05:33:48 [19320, WinNUT]: Update Icon
2024/6/29 星期六 下午 05:33:49 [19320, WinNUT]: Main Gui Lose Focus
2024/6/29 星期六 下午 05:33:49 [19320, WinNUT]: Update Icon
2024/6/29 星期六 下午 05:33:49 [19320, Pref_Gui]: Show LogFile
```

Thanks again for your help.

gbakeman, 2024-06-29:
Could you please verify for me that IP address of the computer running WinNUT has been added to the Permitted Diskstation Devices list on your Synology? (Please see [here](https://github.com/gawindx/WinNUT-Client/issues/47#issuecomment-759180793) for more info.)

timmyhbk, 2024-06-29:
Yes, my settings are consistent with that article.

gbakeman, 2024-06-29:
Going back to when you said "the GUI doesn't change", do the gauges remain in a frozen state as if you were still connected to the UPS, even though you aren't? I've had issues with the gauges losing their state during a disconnection, although the rest of this is odd... I'm not really seeing the output I'd expect in your logs, so I'm going to need to ask you to do a slightly more involved test for me.

First thing, please open the settings in WinNUT and make sure log output is set to Debug. Afterwards, hit the Ok button and exit WinNUT. Then navigate to the log files located at `C:\Users\{ username }\AppData\Roaming\NUTDotNet\WinNUT Client\2.3.8892.21649\` and clear out all current log files.

Now, please restart WinNUT and do the following things for me:

1. Connect if it's not already
2. Open and close UPS Variable dialog
3. Suspend/sleep your system
4. Wait for about 5 seconds after it's suspended
5. Resume operation again
6. Open and close UPS Variable dialog again
7. Exit WinNUT

After doing all of that, please directly attach the log file to your next comment here. Don't worry about translating anything, although I appreciate you doing that for me. And thank you for working with me to troubleshoot this issue.

timmyhbk, 2024-06-29:
>Going back to when you said "the GUI doesn't change", do the gauges remain in a frozen state as if you were still connected to the UPS, even though you aren't? I've had issues with the gauges losing their state during a disconnection, although the rest of this is odd...

I've noticed that if I put Windows to sleep and wake it up within a short period of time, I'm also experiencing gauge loss.
And when there is a longer interval between sleep and wake, the gauge freezes in its state before sleep.

This is the Log after following the instructions:

```
2024/6/30 星期日 上午 12:34:20 [22196, Logger]: WinNUT Client 2.3.8892.21649 Log file init
==== History of 21 previous events ====
[1] 2024/6/30 星期日 上午 12:34:20 [22196, MyApplication]: MyApplication_Startup complete.
[2] 2024/6/30 星期日 上午 12:34:20 [22196, WinNUT]: NotifyIcons Initialised
[3] 2024/6/30 星期日 上午 12:34:20 [22196, WinNUT]: Windows 10 Toast Notification Not Available. Required Version: 10.0.18362.0, Current: 6.2.9200.0
[4] 2024/6/30 星期日 上午 12:34:20 [22196, WinNUT]: Windows App Use Dark Theme
[5] 2024/6/30 星期日 上午 12:34:20 [22196, WinNUT]: Windows Use Dark Theme
[6] 2024/6/30 星期日 上午 12:34:20 [22196, WinNUT]: NotifyIcon Text => 
WinNUT Client - 2.3
未連線
[7] 2024/6/30 星期日 上午 12:34:20 [22196, WinNUT]: Status Icon Changed
[8] 2024/6/30 星期日 上午 12:34:20 [22196, WinNUT]: New Icon Value For Systray : 1216
[9] 2024/6/30 星期日 上午 12:34:20 [22196, WinNUT]: New Icon Value For Gui : 1216
[10] 2024/6/30 星期日 上午 12:34:20 [22196, WinNUT]: Update Icon at Startup
[11] 2024/6/30 星期日 上午 12:34:20 [22196, WinNUT]: Beginning ApplyApplicationPreferences subroutine.
[12] 2024/6/30 星期日 上午 12:34:20 [22196, WinNUT]: Initializing all display values and configurations.
[13] 2024/6/30 星期日 上午 12:34:20 [22196, WinNUT]: Parameter Dial Input Voltage Need to be Updated
[14] 2024/6/30 星期日 上午 12:34:20 [22196, WinNUT]: Parameter Dial Input Voltage Updated
[15] 2024/6/30 星期日 上午 12:34:20 [22196, WinNUT]: Parameter Dial Input Frequency Need to be Updated
[16] 2024/6/30 星期日 上午 12:34:20 [22196, WinNUT]: Parameter Dial Input Frequency Updated
[17] 2024/6/30 星期日 上午 12:34:20 [22196, WinNUT]: Parameter Dial Output Voltage Need to be Updated
[18] 2024/6/30 星期日 上午 12:34:20 [22196, WinNUT]: Parameter Dial Output Voltage Updated
[19] 2024/6/30 星期日 上午 12:34:20 [22196, WinNUT]: Parameter Dial Voltage Battery Need to be Updated
[20] 2024/6/30 星期日 上午 12:34:20 [22196, WinNUT]: Parameter Dial Voltage Battery Updated
[21] 2024/6/30 星期日 上午 12:34:20 [22196, Logger]: WinNUT Client 2.3.8892.21649 Log file init
==== Begin Live Log ====
2024/6/30 星期日 上午 12:34:20 [22196, WinNUT]: WinNut Preferences Applied.
2024/6/30 星期日 上午 12:34:20 [22196, String]: New Log to CB_Current Log : WinNUT 設定已變更
2024/6/30 星期日 上午 12:34:20 [22196, WinNUT]: Run Automatic Update
2024/6/30 星期日 上午 12:34:20 [22196, Update_Gui]: Verify Update
2024/6/30 星期日 上午 12:34:20 [22196, WinNUT]: WinNUT Client v2.3.8892.21649 completed initialization.
2024/6/30 星期日 上午 12:34:20 [22196, WinNUT]: Main GUI activated.
2024/6/30 星期日 上午 12:34:20 [22196, WinNUT]: Update Icon
2024/6/30 星期日 上午 12:34:20 [22196, WinNUT]: WinNUT_Shown for the first time.
2024/6/30 星期日 上午 12:34:20 [22196, WinNUT]: Update Icon
2024/6/30 星期日 上午 12:34:20 [22196, WinNUT]: Minimize WinNut On Start
2024/6/30 星期日 上午 12:34:20 [22196, WinNUT]: Update Icon
2024/6/30 星期日 上午 12:34:20 [22196, WinNUT]: Minimize Main Gui To Notify Icon
2024/6/30 星期日 上午 12:34:20 [22196, WinNUT]: Main Gui Lose Focus
2024/6/30 星期日 上午 12:34:20 [22196, WinNUT]: Update Icon
2024/6/30 星期日 上午 12:34:20 [22196, WinNUT]: Auto-connecting to UPS on startup.
2024/6/30 星期日 上午 12:34:20 [22196, WinNUT]: Client UPS_Connect subroutine beginning.
2024/6/30 星期日 上午 12:34:20 [22196, UPS_Device]: Beginning connection: upsmon@192.168.1.101:3493, Name: ups [AutoReconnect]
2024/6/30 星期日 上午 12:34:20 [22196, Nut_Socket]: Attempting TCP socket connection to 192.168.1.101:3493...
2024/6/30 星期日 上午 12:34:20 [22196, Nut_Socket]: Connection established and streams ready for 192.168.1.101:3493
2024/6/30 星期日 上午 12:34:20 [22196, Nut_Socket]: Attempting authentication...
2024/6/30 星期日 上午 12:34:20 [22196, Nut_Socket]: Error while attempting to log in: INVALIDARGUMENT (ERR INVALID-ARGUMENT)
Query: LOGIN
2024/6/30 星期日 上午 12:34:20 [22196, Nut_Socket]: NUT server reports VER: DSM7-2-1-NewModel-repack-64570-230831 NETVER: 1.2
2024/6/30 星期日 上午 12:34:20 [22196, UPS_Device]: Retrieving basic UPS product information...
2024/6/30 星期日 上午 12:34:20 [22196, UPS_Device]: ups.serial is not supported by server.
2024/6/30 星期日 上午 12:34:20 [22196, UPS_Device]: Apply Fallback Value when retrieving ups.serial
2024/6/30 星期日 上午 12:34:20 [22196, UPS_Device]: ups.firmware is not supported by server.
2024/6/30 星期日 上午 12:34:20 [22196, UPS_Device]: Apply Fallback Value when retrieving ups.firmware
2024/6/30 星期日 上午 12:34:20 [22196, UPS_Device]: Initializing other well-known UPS variables...
2024/6/30 星期日 上午 12:34:20 [22196, UPS_Device]: output.current is not supported by server.
2024/6/30 星期日 上午 12:34:20 [22196, UPS_Device]: output.voltage: 111
2024/6/30 星期日 上午 12:34:20 [22196, UPS_Device]: output.realpower is not supported by server.
2024/6/30 星期日 上午 12:34:20 [22196, UPS_Device]: Determining best method to calculate power usage...
2024/6/30 星期日 上午 12:34:20 [22196, UPS_Device]: ups.realpower is not supported by server.
2024/6/30 星期日 上午 12:34:20 [22196, UPS_Device]: Using RPNomLoadPct method.
2024/6/30 星期日 上午 12:34:20 [22196, UPS_Device]: battery.capacity is not supported by server.
[... cut ...]
2024/6/30 星期日 上午 12:35:24 [22196, WinNUT]: New Icon Value For Gui : 1216
2024/6/30 星期日 上午 12:35:24 [22196, WinNUT]: NotifyIcon Text => 
WinNUT Client - 2.3
未連線
2024/6/30 星期日 上午 12:35:24 [22196, WinNUT]: Battery Status => Unknown
2024/6/30 星期日 上午 12:35:24 [22196, WinNUT]: Disconnected from Nut Host
2024/6/30 星期日 上午 12:35:24 [22196, String]: New Log to CB_Current Log : 已與 NUT 伺服器中斷連線。
2024/6/30 星期日 上午 12:35:24 [22196, WinNUT]: Main Gui Lose Focus
2024/6/30 星期日 上午 12:35:24 [22196, WinNUT]: Update Icon
```

gbakeman, 2024-06-29:
Thank you once again for doing that. I think I'm starting to understand what the issue may be - the Synology version of NUT server may be unhappy that WinNUT disappears without doing the LOGOUT command, and then doesn't accept any commands following that. Let me wrap up something I'm working on, then I'll take a deeper dive into this.

gbakeman, 2024-07-01:
Just had a chance to dig into this. To be honest, I'm shooting in the dark a little bit, but I picked what looked like an obvious low-hanging fruit; implementing basic support for system suspend events. There's a debug build up [here](https://github.com/nutdotnet/WinNUT-Client/actions/runs/9750200053?pr=163), please give it a try and let me know how it goes. Thanks for your patience and testing efforts!

timmyhbk, 2024-07-02:
After several tests. The first few times I woke up after a short sleep (about 5sec), the connection was successfully restored. However, after extending the interval between sleep and wakeup (5+mins), the connection still can't be resumed. During this time, I encountered several crashes.

Strangely. Now I can't restore the connection after sleep in any way :(

Thank you!

**Crash Report:**
[CrashReport_2024-07-02T14.13.13.txt](https://github.com/user-attachments/files/16067559/CrashReport_2024-07-02T14.13.13.txt)
[CrashReport_2024-07-02T18.13.09.txt](https://github.com/user-attachments/files/16067560/CrashReport_2024-07-02T18.13.09.txt)
[CrashReport_2024-07-02T19.51.41.txt](https://github.com/user-attachments/files/16067561/CrashReport_2024-07-02T19.51.41.txt)

**Some Log**
[WinNUT-Client-2024-07-02(5+).log](https://github.com/user-attachments/files/16067586/WinNUT-Client-2024-07-02.5%2B.log)
[WinNUT-Client-2024-07-02(5sec).log](https://github.com/user-attachments/files/16067587/WinNUT-Client-2024-07-02.5sec.log)
[WinNUT-Client-2024-07-02(15+).log](https://github.com/user-attachments/files/16067588/WinNUT-Client-2024-07-02.15%2B.log)

gbakeman, 2024-07-02:
Wow, that is not what I was expecting! 😅

Your detailed logs have led me to a deeper dive in WinNUT, and I've discovered several issues and cleaned up some dusty code that should have it handling connections better. I also discovered that it was using the NUT `LOGIN` command incorrectly, so WinNUT will now Login to the UPS properly. Please go ahead and [test the latest debug](https://github.com/nutdotnet/WinNUT-Client/actions/runs/9767950177?pr=163) build again. I want to let you know that I'll be leaving in two days and won't be around for a couple of weeks, so I may disappear. But hopefully we'll have this fixed soon.

timmyhbk, 2024-07-03:
The new version still encounters the same situation. Sometimes it works. But most of the time the connection is still not restored. And sometimes it crashes.

**Log:**
[WinNUT-Client-2024-07-03-1.log](https://github.com/user-attachments/files/16084715/WinNUT-Client-2024-07-03-1.log)
[WinNUT-Client-2024-07-03-2.log](https://github.com/user-attachments/files/16084718/WinNUT-Client-2024-07-03-2.log)
[WinNUT-Client-2024-07-03-3.log](https://github.com/user-attachments/files/16084719/WinNUT-Client-2024-07-03-3.log)

**Crash Report:**
[CrashReport_2024-07-03T13.33.58.txt](https://github.com/user-attachments/files/16084769/CrashReport_2024-07-03T13.33.58.txt)
[CrashReport_2024-07-03T16.23.14.txt](https://github.com/user-attachments/files/16084770/CrashReport_2024-07-03T16.23.14.txt)

I tried deleting user.config from appdata and resetting it. At first it appeared differently than before. It kept trying to reconnect 30 times. But in the end it still could not reconnect. But after I switched the log level to debug it was not reproducible. (Even if I change it back to Notice)

**Here is the log for that time:**
[WinNUT-Client-2024-07-03(Notice).log](https://github.com/user-attachments/files/16084854/WinNUT-Client-2024-07-03.Notice.log)

gbakeman, 2024-07-05:
And thank you for your patience while we get to the bottom of this! I'm glad to hear that the crashes seem to have been fixed at least. I'm currently away now so we'll have to take a break, but I'll be back on this in the second half of the month.

gbakeman, 2024-07-15:
Just got back and had a chance to take another look at this. For whatever reason, Synology's NUT server does not want to share it's version with you after the client comes back from sleep (access denied error). I'm still have to wonder if this is because we're not disconnecting gracefully when going to sleep, but [here's another build](https://github.com/nutdotnet/WinNUT-Client/actions/runs/9945861245?pr=163) where I make WinNUT more fault-tolerant to servers that don't like sharing their version information. No need to run multiple tests if you still see a problem, just one log for each test is fine. I am curious though, if you do see issues, I'd like for you to delete your username and password from the preferences of WinNUT. This will stop it from trying to login, which may make the Synology happy when the client disconnects unexpectedly. Thanks again for your patience.

gbakeman, 2024-07-18:
There's a 2-3 minute lag before WinNUT recognizes it's disconnected and tries reconnecting? If you'd like to troubleshoot that further, could you upload a log of that session?

timmyhbk, 2024-07-18:
I don't know why the previous log was not recorded. When I try to log, the recovery interval is not so long.
[WinNUT-Client-2024-07-18.log](https://github.com/user-attachments/files/16283845/WinNUT-Client-2024-07-18.log)

But anyway, this result is good enough for me. Thanks for your help!

gbakeman, 2024-07-18:
Alright, hopefully just a fluke. And thanks again for all of your efforts with testing and providing feedback. I'll have this merged and closed in a little bit, then a pre-release update will be out shortly.

gbakeman, 2024-07-18:
Just before I close this - I took a look through your last log, and it appears that every variable is failing when you reconnect to your NUT server. It definitely seems like the Synology NUT server is behaving strangely. If you want, feel free to open a new discussion or issue and we can try to further troubleshoot what's going on.

closed 2024-07-18 by gbakeman

dwd1, 2025-10-17:
Unfortunately the problem is not solved. After wake from sleep the WinNUT doesnt reconnect,

gbakeman, 2025-10-17:
Hi @dwd1,

Would you please [create a new Q&A post](https://github.com/nutdotnet/WinNUT-Client/discussions/new?category=q-a) and I'll see how I can help? When making the post, I'd like for you to upload the log file (set to Debug in settings first).

dwd1, 2025-10-18:
Sometimes it happens, sometimes it doesn't. I turned on the debug log and if it happens again I will upload it. Thanks!

dwd1, 2025-11-11:
Where is the logfile located?
When I click on the (i) icon it opens only the EditPlus editor not the file.

gbakeman, 2025-11-11:
WinNUT's logs are stored in your user's local AppData folder, under a NUTDotNet folder, then a few more levels below that for the client and version. The button from earlier should open the file directly in an editor which should have the exact path printed.

gbakeman, 2025-11-26:
If you want to make a VM to test yourself that could be helpful, I was just saying that I already have one so I may be able to do some testing myself at some point. But let's continue discussion in a new topic to gather more information.