nutdotnet/WinNUT-Client #167: "Critical Error occured in WinNUT" in pre-release v2.3.8965

Hedders, 2024-07-18:
### Discussed in https://github.com/nutdotnet/WinNUT-Client/discussions/166

<div type='discussions-op-text'>

<sup>Originally posted by **Hedders** July 18, 2024</sup>
Hello, just installed the new v2.3.8965 and it fails to launch, displaying a broken-looking dialog with the error "Critical Error occurred in WinNUT".

I'm on Windows 11 23H2, build 22631.3880, with .NET framework 4.8.09032. The Windows event log shows event ID 1000 (Application Error), with the following diagnostic information:

Faulting application name: WinNUT-Client.exe, version: 2.3.8965.28008, time stamp: 0x66996e11
Faulting module name: KERNELBASE.dll, version: 10.0.22621.3880, time stamp: 0xee1eae1c
Exception code: 0xe0434352
Fault offset: 0x0014a782
Faulting process ID: 0x0x1D34
Faulting application start time: 0x0x1DAD9547DC0AC6E
Faulting application path: C:\Program Files (x86)\WinNUT-client\WinNUT-Client.exe
Faulting module path: C:\WINDOWS\System32\KERNELBASE.dll
Report ID: 77f6d5b7-99a6-42cd-9d82-4e7e13de4fdd
Faulting package full name: 
Faulting package-relative application ID: 

Immediately prior to that in the event log is a .NET Runtime Event 1026, with the following diagnostic information:

Application: WinNUT-Client.exe
Framework Version: v4.0.30319
Description: The process was terminated due to an unhandled exception.
Exception Info: System.MissingMethodException
   at WinNUT_Client.Controls.UPSVarGauge.set_Value1(Single)
   at WinNUT_Client.WinNUT.InitializeComponent()
   at WinNUT_Client.WinNUT..ctor()

Exception Info: System.InvalidOperationException
   at WinNUT_Client.My.MyProject+MyForms.Create__Instance__[[System.__Canon, mscorlib, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b77a5c561934e089]](System.__Canon)
   at WinNUT_Client.My.MyApplication.MyApplication_UnhandledException(System.Object, Microsoft.VisualBasic.ApplicationServices.UnhandledExceptionEventArgs)
   at Microsoft.VisualBasic.ApplicationServices.WindowsFormsApplicationBase.raise_UnhandledException(System.Object, Microsoft.VisualBasic.ApplicationServices.UnhandledExceptionEventArgs)
   at Microsoft.VisualBasic.ApplicationServices.WindowsFormsApplicationBase.OnUnhandledException(Microsoft.VisualBasic.ApplicationServices.UnhandledExceptionEventArgs)
   at Microsoft.VisualBasic.ApplicationServices.WindowsFormsApplicationBase.DoApplicationModel()
   at Microsoft.VisualBasic.ApplicationServices.WindowsFormsApplicationBase.Run(System.String[])
   at WinNUT_Client.My.MyApplication.Main(System.String[])

Any suggestions gratefully received!</div>

Fr0stX76, 2024-07-18:
The no installer works fine, si its the msi

gbakeman, 2024-07-18:
Hi all,

Critical mistake on my part - I didn't test the MSI upgrade before deploying. Sorry!

Because I replaced the AGauge library in WinNUT, the new one actually has a lower version than the one that was deployed with WinNUT's version (2.3 vs 2.1.) It seems like the MSI decided it didn't need to upgrade the previous one, leaving the installation with a mismatched library (crash.) Please uninstall this version of WinNUT and reinstall from the latest pre-release currently available.

I need to work on the AGauge project a little more and rebrand the library so the installer knows to replace it next time.

Hedders, 2024-07-19:
Thank you for responding so swiftly!

gbakeman, 2024-07-23:
All,

Thanks for reporting this so quickly. A new pre-release is out, and from my own testing, it seems to be updating correctly! Please let me know if there are any further issues.

closed 2024-07-23 by gbakeman

Hedders, 2024-07-23:
Thank you! Will test shortly and report back.

Hedders, 2024-07-23:
Yep, can confirm all is copacetic.  Thank you!

gbakeman, 2024-07-23:
Great, thank you for the update!