nutdotnet/WinNUT-Client #182: Connected pop up not appearing after reconnecting in some situations

gbakeman, 2024-08-23:
> On Win10 I got a single notification when the nut server stopped. "Reconnection In Progress\nReconnection..."
> To be perfect a second notification an hour later when the nut server restarted would be nice.
> 
> When I look at the recent messages drop down list in WinNUT Client there are three messages.
> 
> 1. Lost Connect to rpi:3493  (should it say "Lost Connect**ion** to rpi:3493"?)
> 2. Reconnection In Progress  (should the notification say "Lost Connection: Reconnection In Progress" as no "Connection lost" notification was given?)
> 3. Connection to Nut Host rpi:3493 Established (should this GUI message also be sent as a Windows notification?)
> 
_Originally posted by @yoyoma2 in https://github.com/nutdotnet/WinNUT-Client/issues/65#issuecomment-2267559170_

yoyoma2, 2024-09-24:
If I stop nut-server on the rpi and start it a minute later I see this in the Win-NUT logs after the connection is reestablished. The last line saying "UPS statuses have not changed since last update, skipping" is suspicious.
```
9/24/2024 12:31:01 PM [3344, WinNUT]: NotifyIcon Text => 
WinNUT Client - 2.3
Connected
9/24/2024 12:31:01 PM [3344, WinNUT]: Connection to Nut Host Established
9/24/2024 12:31:01 PM [3344, String]: New Log to CB_Current Log : Connection to Nut Host rpi:3493 Established
9/24/2024 12:31:01 PM [3344, UPS_Device]: Nut Host Reconnected
9/24/2024 12:31:06 PM [3344, UPS_Device]: Enter Retrieve_UPS_Datas
9/24/2024 12:31:06 PM [3344, UPS_Device]: input.frequency is not supported by server.
9/24/2024 12:31:06 PM [3344, UPS_Device]: Apply Fallback Value when retrieving input.frequency
9/24/2024 12:31:06 PM [3344, UPS_Device]: output.voltage is not supported by server.
9/24/2024 12:31:06 PM [3344, UPS_Device]: Apply Fallback Value when retrieving output.voltage
9/24/2024 12:31:06 PM [3344, UPS_Device]: UPS statuses have not changed since last update, skipping.
```
I cannot see in UPS_Device.vb where variable oldStatusBitmask gets changed when Socket_Broken happens. Shouldn't some sort of flag be set so that the notification of the connection being reestablished is not skipped?

gbakeman, 2024-09-24:
Hi,

Thanks for giving the latest build another try and providing feedback! And also, thanks for diving in to the code to see if you could find out what's wrong.

With regards to the status updating, I don't think that's a big concern since [this](https://github.com/nutdotnet/WinNUT-Client/blob/ca65848a1d13ebeb6166dc508582da1557d29602/WinNUT_V2/WinNUT-Client/WinNUT.vb#L960) is the only method that acts on the status changing, which is only to handle a few cases of the UPS switching between line power and battery, as well as a full shutdown signal. I'm not sure I see any reason to clear out/reset the old status bitmask during a reconnection event, but please let me know if I'm missing something.

I think the log data you provided may have cut out the new logging statements I had added to the [ToastNotifyIcon](https://github.com/nutdotnet/WinNUT-Client/blob/ca65848a1d13ebeb6166dc508582da1557d29602/WinNUT_V2/WinNUT-Client/WinNUT.vb#L869) method, which I think still should've run. Could you please upload more of the log in case those messages are still there?

yoyoma2, 2024-09-24:
I'm running the latest pre-release v2.3.8988 not the debug build. Machines on a UPS are important so a pre-release is already pushing it and a debug build is not possible. Is the extra logging code merged?

I don't know the Win-NUT code at all...

If you stop the NUT server and start it again 1 min later do you get actually get a Windows notification from Win-NUT when the connection is lost? Do you get actually get a Windows notification from Win-NUT when the connection is restored? I don't think my setup is special. I only get a Reconnection In Progress notification when the connection is lost.

gbakeman, 2024-09-24:
It is odd, I've tried simulating the server going down by disabling the network adapter on my PC, but maybe there's some special nuance to actually shutting down the NUT server. I just can't seem to reproduce the issue you've been facing, so I was hoping the extra logging I'd added might help in getting there.

I totally understand the desire for stable software and how it would be uncomfortable running anything else for an extended period of time. If you could consider running this debug build just for a few minutes to grab a log file though, I think that could potentially help here. And sorry, I completely forgot to link it! If this is something you.could do, here's the build in question: https://github.com/nutdotnet/WinNUT-Client/actions/runs/10531599982

yoyoma2, 2024-09-25:
I installed Visual Studio Community and hacked something that made the "Connected" notification appear when the reconnection occurs. I added a log message if we are logging in twice and it was showing up in the log file so I tried a hack to avoid logging in again while reconnecting and it works.

I don't know the code so there must be a better solution to handle the thrown exception but this should provide useful information as to what is happening.

```
$ git diff WinNUT-Client_Common/UPS_Device.vb
diff --git a/WinNUT_V2/WinNUT-Client_Common/UPS_Device.vb b/WinNUT_V2/WinNUT-Client_Common/UPS_Device
.vb
index 88595e7..ceeb847 100644
--- a/WinNUT_V2/WinNUT-Client_Common/UPS_Device.vb
+++ b/WinNUT_V2/WinNUT-Client_Common/UPS_Device.vb
@@ -159,7 +159,12 @@ Public Class UPS_Device
     End Sub

     Public Sub Login()
+        If IsReconnecting = True Then
+            Return
+        End If
+
         If Not IsConnected OrElse IsLoggedIn Then
+            LogFile.LogTracing("UPS is in an invalid state to login.", LogLvl.LOG_ERROR, Me)

             Throw New InvalidOperationException("UPS is in an invalid state to login.")
         End If
```

gbakeman, 2024-09-25:
It's possible the socket isn't resetting its logged in state when disconnected. I'll take a closer look later today and hopefully have another debug build up for you soon.

gbakeman, 2024-09-25:
No worries at all about the questions, I'm glad you're curious and I appreciate everything you've done so far.

A little background about logging in to NUT servers - it's actually not necessary a vast majority of the time and should only be done under specific circumstances. I'm tracking this info in #106 if you want a closer look. I made some big changes with connection code recently in #163 so that the Login command started being correctly called, and made it so that it's only called in the main WinNUT connection routine as long as the provided username is not empty. This is why I'm confused that your fix worked; the Login attempt is only made at the very end of the connected event logic (so the notifications should have been triggered already), and it shouldn't have interrupted anything anyways. Looking back at your last full log, the point where the notificaiton toast was supposed to appear still shows even though it looks like the client didn't attempt to login when your RPi came back up (also a mystery...)

> BTW the "Connected" toast debug log messages didn't appear in the case where the exception was thrown on double login. With the hack to avoid double login, the toast debug log messages appear (as do the Windows notifications).

That's very odd to me. I want to try testing with my own VM instance of NUT running to fully emulate your situation, although I may not be able to get to that for a few days. You also mention this exception which isn't present in your last full log as well - if you could please upload a debug log of the build linked above for me like you were doing before, that should help me get closer to the root of the issue. Thanks again for everything, and please keep asking more questions!

yoyoma2, 2024-09-26:
I've cleared the "Login" text field in the latest pre-release which I use in "production" and the "Connected" notification now appears when the NUT server is restarted. I'll leave it that way for now but I'm still very interested in testing your next build.

The additional log message above (or similar) that I added to the Login subroutine may be a good idea as it would have made identifying that this was login related much easier.

gbakeman, 2024-09-26:
I'd really appreciate it if you could upload a recent complete log file with this exception you've been mentioning as that will help me a ton.

yoyoma2, 2024-09-26:
Here's my only debug log from my build that includes **one** "UPS is in an invalid state to login." message that I added. I then immediatelty added the hack so this exception message never happened again. Although my git pull was from the main branch I had manually added your pending pull request so the toast debug stuff is in there in addition to my change.

[WinNUT-Client-2024-09-25.zip](https://github.com/user-attachments/files/17149513/WinNUT-Client-2024-09-25.zip)

Hope this helps.

gbakeman, 2024-09-26:
That helped a lot, thank you! I was finally able to reproduce the issue you were experiencing. It's possible my client wasn't even logging in on my other NUT server which explains why I wasn't seeing the issue. Now that I've setup a test NUT server for logging in, I've been able to understand that process a little better.

It was pretty much exactly as you thought. An exception was being thrown because WinNUT thought it was already logged in. When that happened, further event handlers in WinNUT (like the notification toast) were skipped. Your hack worked because it avoided the exception like you were thinking. I've approached the problem a little differently, in that I have WinNUT properly unset the logged in flag when disconnecting so it will always sync up now and allow WinNUT to log back in when reconnecting. And from what I can tell, the NUT server doesn't care how the client disconnects, so you can always log back in again.

If you could please, I'd like for you to once again test the [newest debug build](https://github.com/nutdotnet/WinNUT-Client/actions/runs/11059562974?pr=183) just to confirm that it works as expected. No need to provide any log info if everything is good. Once you can confirm for me, I'll merge and close this issue, then hopefully have a new pre-release up soon.

Thanks again for all of your assistance with this, I'm sorry it ended up being such a simple problem!

yoyoma2, 2024-09-27:
Not easy to notice a failed login when things work anyways without logging in...

It works beautifully! I get the "Connected" notification at the right time and I even checked in the logs that the authentication is successful just before the reconnected message.
```
09/26/2024 20:03:12 [18028, Nut_Socket]: Authenticated successfully.
09/26/2024 20:03:12 [18028, UPS_Device]: Nut Host Reconnected
```
It's cool when everything makes sense in the end. I'll use the pre-release build in production when available and reinsert the previously cleared login name. Maybe leave a comment here about the new build when available so I get an email.

Very pleasant cooperating with you.

gbakeman, 2024-09-27:
As I merged these changes, I realized that I have the main development branch half-way through implementing a new deployment system for WinNUT. If you can hang on for a little bit longer, I'd like to get this out of the way so future updates are easier.

yoyoma2, 2024-09-28:
No rush. The latest pre-release is doing the job in _not_ logged in mode. When the next pre-release comes out I'll try the logged in mode to be consistent with my linux NUT clients and to test your fix more thoroughly.

yoyoma2, 2025-06-03:
Just like @kirsch33 in issue #65 I'm still successfully running WinNUT but the pre-release not the debug build. I have the "Login" text field empty thus avoiding the double login bug. I'd certainly appreciate a new pre-release build.

yoyoma2, 2025-11-05:
Maybe something strange in my setup. I'll reproduce it and provide a debug log as soon as possible. A day or two hopefully.

gbakeman, 2025-11-06:
At this point, I have WinNUT configured to connect to my VM NUT server, logging disabled (but still set to Debug), and minimized to tray. I feel like I've gotten almost everything replicated here except for the reboot (and start at boot) part of it - but still can't reproduce 😕. Just to be clear, you allow the PC to boot up and have WinNUT establish its connection, then bring down the NUT server? Maybe the launching WinNUT at startup has something to do with it? I've been meaning to spin up a full Windows client VM with WinNUT inside so I can test things like rebooting and suspends, so that might be a worthwhile project to tackle next. I'm sorry I'm struggling so much, I know this is a fairly big bug all things considered so it's important to fix. Thank you for your hard work in helping me diagnose this. If you have any other ideas as to what I'm missing here, I'm all ears!

gbakeman, 2025-11-06:
> After it works with logging on, it will probably still work if logging is turned off. Sometimes after the bug I'll disconnect and reconnect manually from the connection menu so the dials can display properly again. After that the bug may or may not be reproducible. That's why I put reboot in the replication steps earlier because after rebooting with logging unchecked the bug always happens. Seems like an initialization thing.

Okay, that's a very interesting observation. I just got a Windows VM spun up, and installed the latest pre-release of WinNUT. Thanks to your observations, I've managed to reproduce the issue. What I've isolated it down to at this point is: so long as a text file of any name (*.txt) is present inside the folder where log files normally go, and logging is turned off, then WinNUT doesn't detect the connection going down. I'm not sure why rebooting or manually disconnecting/reconnecting seems to affect it, but it's definitely some kind of initialization issue like you said.

> Even though I don't understand the code, last year I was able to build from source and at least see that a double login attempt was throwing an exception. That was enough information for you to come up with a fix. Would pulling the latest from the dev-2.3 give me the proper code to test with? Hopefully it happens in the debugger.

I don't think there's any need for you to dive in with the debugger at this point, but thank you for offering. (Although if you're interested generally, I try to keep the current default branch (dev-2.3 in this case) as a reliable starting point for anyone who wants to contribute.) Let me investigate this a little more, open a new issue on your behalf and we can go from there.

Thanks again for all of your assistance and input on this.

gbakeman, 2025-11-07:
@yoyoma2 In case GitHub didn't ping you, I've opened #223 to track this new issue if you'd like to follow along and contribute. I'll close this current issue for now.