nutdotnet/WinNUT-Client #195: WinNUT shuts down immediately when remaining battery time is "Unavailable", even if remaining time trigger set to 0

GeraZsolt, 2024-11-21:
- [X ] Searched to see if this issue exists
- [X ] Completed below form
- [ ] Attached debug log file
- [ ] Copy output in File -> UPS Variable

**WinNUT Version: ** 
v2.3.8988

**Windows OS Version: ** 
Win10 22H2

**Describe the bug**
My UPS (SPS-MID 1000 OnLine UPS) battery remaining time is unknown on the Synology DS420+ NAS, so the info transmitted to WinNUT Client also can not read the remaining time, so it is "Unavailable", and even if i set the "Shutdown if runtime lower than (sec)" is set to 0 (ZERO) the shutdown starts immediately.
ZERO value should disable the check for remaining battery runtime!

OR

There should be a logical operator to be able to set this one:
 "Shutdown if battery lower than x%" OR "Shutdown if runtime lower than (sec)" will be the trigger (whichever comes true first)
or this one:
 "Shutdown if battery lower than x%" AND "Shutdown if runtime lower than (sec)" will be the trigger if BOTH of them are true.

ZERO value disables the check for remaining battery percent, or battery remaining time.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'settings / shutdown options'
2. Set runtime to '0'
3. Test power outage
4. Shutdown will be initiated when the UPS runtime is unavailable.

**Expected behavior**
I expect that WinNUT ignores the unavailable remaining time if i set the trigger to 0 sec. (and also the same for the remaining percent, if a UPS can not communicate this attribute to the SNMP server)

**Screenshots**

**Additional context**

gbakeman, 2024-11-27:
Hi there,

Sorry for taking awhile to get back to you. I want to say that it was intended for zero values to disable those checks, however I realize now that I may have introduced a slight bug here. WinNUT was designed originally such that when a variable was unavailable from the UPS, WinNUT would substitute the variable for a nominal value (so battery voltage would be set to 12V if it was unavailable, for example.) See #116. This probably worked fine in your scenario because the time remaining or percentage values would be nominal, thus effectively disabling those checks. However, I intentionally made it so that unavailable variables were set to out of range values, like 0 or -1. So I see now how this rendered the checks ineffective whenever a variable is missing.

I'm going to have to think how to approach this one a little bit (and your feedback is always appreciated.) On the one hand, I do want to fix the underlying system of how we handle variables in WinNUT - ideally, if a variable is unavailable, we'd just mark it as such. This could potentially indicate a need for me to accomplish an even greater project in #132 which is to transition the client code in WinNUT to a new library I've been working on. That could take quite a lot of work, however. On the other hand, I also want to overhaul how WinNUT handles events (see #92) and create a user-customizable system so people can create custom event handlers for different events.

I may have to think about this one for a little. In the meantime, does the Delay to Shutdown option let you at least delay the shutdown?

GeraZsolt, 2024-11-29:
Thank You for the response! :)
I appreciate the long explaining! :)
Hope Your plans come true! :)
You asked about the workaround, and yes, the delay works, and holds back WinNUT from shutdown the PC as soon as the UPS lost power from the wall.
Have a great winter and coding! :)

Zsolt

SebastianOehm, 2025-10-19:
@gbakeman This is still having the same behaviour even after almost a year. But if it is so difficult finding a right approach, why not simply introduce a checkbox to ignore "time available?"

gbakeman, 2025-10-19:
Hi @SebastianOehm,

Unfortunately a lot of other things have come up, and besides me getting sidetracked on a translation-related project for WinNUT I haven't been dedicating much time to other things that are affecting people so I have to apologize for that.

Looking back at this bug, I want to say that the _simplest_ method for fixing it would be to modify [this part](https://github.com/nutdotnet/WinNUT-Client/blob/c556b7829f1480c7bf18abef3e1b7105e7a42140/WinNUT_V2/WinNUT-Client/WinNUT.vb#L641) of the code (note to future self) such that, if either of the two variables is set to -1 (indicating unavailable), WinNUT will not process that specific condition for powering down. Right now, my logic assumes that if one of the variables is unavailable, then both must be - which was obviously a faulty assumption.

I can't make any promises, but I do want to fix this bug at some point. Please stay tuned for more info!

gbakeman, 2025-11-02:
Thanks again for your help troubleshooting this issue. [A new Pre-Release build](https://github.com/nutdotnet/WinNUT-Client/releases/tag/v2.3.9437) is out that I hope fixes this. I'll close this assuming all is well, but please comment back if issues persist.

closed 2025-11-02 by gbakeman

SebastianOehm, 2025-11-02:
tested, confirmed working! @gbakeman