nutdotnet/WinNUT-Client #197: WinNUT should always show UPS values even if calibration is set to wrong values

codiflow, 2024-11-27:
**Please make sure to fill out this form completely, and attach your program logs (set to Debug setting). You can delete this line.**

- [X] Searched to see if this issue exists
- [X] Completed below form
- [ ] Attached debug log file
- [ ] Copy output in File -> UPS Variable

WinNUT Version: **v2.3.8988**
Windows OS Version: **Windows Server 2016 Essentials Build 14393**

**Describe the bug**
Some values in the UI are displayed wrong although being provided correctly by the UPS. This has to do with "wrong" calibration settings. The range for battery.voltage was set from 6 to 18 volts but the UPS provided 27.1V. So the UI just displayed garbage values.

Maybe this has to do with some "old settings migration" – I don't know. I'm not sure if this was working before upgrading to the most recent version of WinNUT and I also don't know if the values were calibrated before.

**To Reproduce**
Steps to reproduce the behavior:
1. Set the calibration settings below the settings sent by the UPS
2. Apply
3. See wrong values being displayed in the GUI

**Expected behavior**
The value being provided from the UPS should **always** be visible in the UI – no matter what is set in the calibration settings. The gauge could be exceeded (or maxed out – whatever) but strange / wrong values should never be displayed instead.

**Screenshots**
[image]

[image]

**Additional context**
-

gbakeman, 2024-11-27:
Hi @codiflow ,

Thanks for taking the time to open this issue. I want to direct you towards a few others:

- #24 
- #69 
- #116 
- #136 

It sounds like this issue most closely related to #136 above and could take a little bit of work to do. Do you mind if I have you follow #136 so we can close this one?

codiflow, 2024-11-27:
Sure – I will take care of maintaining my calibration section better in the future but maybe others don't (know that they should)?

The issue at #136 is "the other way round" so there the value is not available and therefore it displays the minimum value. For me the value IS available but it still is displaying the minimum value because the value exceeds the values set in the calibration section.

However you seem to be aware of this issue and others who might search for it here will come across this issue as well. So feel free to close it for now 😎

gbakeman, 2024-11-27:
That's a good point you make too. I will leave this open now for other people to see. And [here](https://github.com/nutdotnet/AGauge/issues/4) is a more specific issue where I intend to solve this problem if it helps at all. Thanks again for your work here!